sparkfun/SparkFun_ADXL345_Arduino_Library #11: Library missing from Arduino IDE

rlogiacco, 2017-09-07:
Please tag the library and have it added to other Arduino libraries by creating an issue in the [Arduino Github repository](https://github.com/arduino/Arduino/issues).

rlogiacco, 2018-02-04:
BUMP
The the required `library.properties` file is present, but this library is still not listed, probably due no issue has been opened on the Arduino Github repository to add this library repository among the watched ones.

nseidle, 2018-02-05:
Thanks for the kicks! This is bad oversight on our part.

Lib manager request: https://github.com/arduino/Arduino/issues/7190
Release in place: https://github.com/sparkfun/SparkFun_ADXL345_Arduino_Library/releases/tag/v1.0.0

Should be a day or less. *Please* do let us know if it doesn't show up in lib manager. Thanks!

closed 2018-02-05 by nseidle